seemoo-lab/nexmon_csi #351: Doubts about 0x88

yjxb1, 2024-06-18:
Can anyone help me answer this question? Why does 0x88 capture QoS frames? I checked the 802.11 protocol framework. The binary of QoS frames is 0010 1000. Shouldn't it be 0x28 when converted to hexadecimal? I don’t quite understand why 0x88 corresponds to QoS frames. Did I understand it wrong?

jlinktu, 2024-06-18:
The Type/Subtype of an IEEE 802.11 QoS Data frame is indeed `0x28`, but in the MAC header of an IEEE 802.11 frame, you are looking at it in the form of the Frame Control Field, which combines Version, Type,  Subtype, and Flags like this:
```
Frame Control Field: 0x8841
    .... ..00 = Version: 0
    .... 10.. = Type: Data frame (2)
    1000 .... = Subtype: 8
    Flags: 0x41
```
Therefore, you want to filter for `0x88` as the first byte, if you want QoS frames only.

yjxb1, 2024-06-19:
> The Type/Subtype of an IEEE 802.11 QoS Data frame is indeed `0x28`, but in the MAC header of an IEEE 802.11 frame, you are looking at it in the form of the Frame Control Field, which combines [...]

Thank you very much for your answer, I still don't understand it. Is the Frame Control Field 0x88 calculated based on Version, Type, Subtype, and Flags? Subtype is converted to hexadecimal 8, but why are Version and Type converted to hexadecimal 8?

jlinktu, 2024-06-19:
Its not a conversion, hexadecimal is just a representation, the data is the same, no matter in what format you look at it.
The combination of Subtype, Type, Version is `b'10001000` (`b'1000` Subtype, `b'10` Type, `b'00 `Version), which in hexadecimal representation is `0x88`.
For more information on how the IEEE 802.11 MAC header is structured, please use the online search engine of your choice, there is plenty of information about it out there. You can also check out the current IEEE 802.11 standard document.

yjxb1, 2024-06-19:
> The combination of Subtype, Type, Version is ( Subtype, Type, Version), which in hexadecimal representation is . For more information on how the IEEE 802.11 MAC header is structured, please use the online search engine of your choice, there is plenty of information about it out there. You can also check out the current IEEE 802.11 standard document.`b'10001000``b'1000``b'10``b'00 ``0x88`

I understand, thank you for answering me in detail

closed 2024-06-19 by jlinktu