seemoo-lab/nexmon_csi #354: fatal error: defs.h no such file or directory

Stefanzcj, 2024-07-11:
when I was doing 'make install-firmware', there is a problem shown in the picture
[image]
How can I solve it

rrathnayaka203, 2024-07-16:
same issue. 🥲 could not sove this.

rrathnayaka203, 2024-07-24:
> > same issue. 🥲 could not solve this.
> 
> Don't try this installation methds anymore. I have tried another one and it works well. The link is: https://github.com/nexmonster/nexmon_csi_bin?tab=readme-ov-file
> 
> I use raspberry pi 4B and the image is: '2022-01-28-raspios-bullseye-armhf'. You can download the image here: https://downloads.raspberrypi.com/raspios_armhf/images/raspios_armhf-2022-01-28/

Thank you very much. This helped.