seemoo-lab/nexmon_csi #359: Installation problems encountered in development

liuchengzhi314159, 2024-09-24:
I had to use a linux kernel 6.1 system during the installation of the csi extension. Because the /lib/module/5.10.21/bulid directory wasn't there in the 5.10 system.
The official guide says it's already supported up to 5.15. but I see the driver is already supported up to 6.6.
However the code I need for the csi extension finds only 5.10 and the driver is in the file. I would like to know if there is any difference between these drivers, i.e. the so-called csi extension can only be supported by the driver of the particular system supported by the extension, or is the extension independent of the particular driver version, in other words, can I implement the installation of the csi extension on a 6.1 driver.
If it is possible to do a csi extension implementation on 6.1, what should I do? The makefiles for the two implementations are a bit hard to read, it feels like the two files are too different.
If there is no way to implement it, is there any step by step guide to tell me how I can pull it up in /lib/module/5.10/bulid now!

jlinktu, 2024-09-24:
https://github.com/seemoo-lab/nexmon_csi/issues/331#issuecomment-1866194237

jlinktu, 2024-09-25:
Yes, that describes one possible way how to do it. Go ahead and try it out.

Regarding python2, just make sure that python in your nexmon environment is python2 not python3. There are many ways to achieve that, please consult the web search engine of your choice.