seemoo-lab/nexmon_csi #361: RaspberryPI OS 5.4.83   make install-firmware  => Makefile:10: *** non-numeric first argument to 'word' function: '-1'. Stop

kuku000, 2024-09-26:
I follow the guideline from https://github.com/seemoo-lab/nexmon_csi.
When I come to step 10, I got an error "Makefile:10: *** non-numeric first argument to 'word' function: '-1'. Stop"

(10. Enter the created subdirectory nexmon_csi and run make install-firmware to compile our firmware patch and install it on the Raspberry Pi )

Raspberrypi 3B+
kernel: 5.4.83

jlinktu, 2024-09-26:
You probably did not setup the build environment by executing `source setup_env.sh` from within the nexmon repositorie's root directory.

jlinktu, 2024-09-26:
The error you are seeing occurs because https://github.com/seemoo-lab/nexmon_csi/blob/fdb25ef0e4e1402e968bb644d4914ad1a3d0a84d/Makefile#L9 produces an empty output. This line splits the content of variable `FW_PATH` based on the delimiter `/`. The variable is read in here: https://github.com/seemoo-lab/nexmon_csi/blob/fdb25ef0e4e1402e968bb644d4914ad1a3d0a84d/Makefile#L2
which in your case should include the following file: [version.mk](https://github.com/seemoo-lab/nexmon/blob/d846e7f48a75695d28167c587ea0b0693577a266/patches/bcm43455c0/7_45_189/version.mk) containing `FW_PATH=$(NEXMON_ROOT)/firmwares/bcm43455c0/7_45_189`.

Check if anything on this way goes wrong for you.

You might have placed `nexmon_csi` in a wrong directory.

kuku000, 2024-09-27:
got it ,thx!!!

closed 2024-09-27 by jlinktu