seemoo-lab/nexmon_csi #364: Raspberry Pi 5 support?

Junye97, 2024-10-15:
Hi there, 

I am wondering if Raspberry Pi 5 support is planned in the development pipeline? To my understanding, RPi 5 uses the same WiFi chip as Pi 4, which should make patching a bit easier. However, Pi 5 runs a much newer kernel than the currently supported 5.10 and also requires a 64-bit kernel. Does Nexmon CSI support 64-bit kernels at all? 

Many thanks

jlinktu, 2024-10-17:
I suggest heading over to https://github.com/seemoo-lab/nexmon , check out the how to compile for 64-bit, adapt the Makefile of `nexmon_csi` to use the location of newer drivers https://github.com/seemoo-lab/nexmon/tree/master/patches/driver or copy the driver of your choice over to your `nexmon_csi` patch and edit the Makefile to choose it. With that you should be able to use it on a Raspberry Pi 5.

Junye97, 2024-10-23:
Thank you very much @jlinktu for the information! I will have a look and see if I can get that to work

closed 2024-10-23 by Junye97

pmssantos, 2025-06-30:
@Junye97 , where you successful in collecting CSI with RPI 5? Thanks

Junye97, 2026-01-28:
> [@Junye97](https://github.com/Junye97) , where you successful in collecting CSI with RPI 5? Thanks

Hi @pmssantos , I haven't got around to work on RPi5 unfortunately...