seemoo-lab/nexmon_csi #368: Is there the information about interpretation of extracted CSI data?

Urb1101, 2024-10-31:
Hello, I dont know structure of CSI data from nexmon extractor. that looks consist of 16 bytes info about each packet, 64bytes offset(this seems to have source MAC address) and payloads(the number of bytes is defined as 'NFFTS' in csireader.m).
In this page, its said that payloads contain int16 imaginary part and int16 real part,
in addition, I think that the first half have real part each subcarrier and the letter half have imaginary part each subcarrier.
I interpreted these by reading csireader.m and plotcsi but I want to know correct structure.
someone know about this or where I can find infomation?

sorry for my bad english and knowledge.

jlinktu, 2024-10-31:
- Header format here: https://github.com/seemoo-lab/nexmon_csi/blob/fdb25ef0e4e1402e968bb644d4914ad1a3d0a84d/src/csi_extractor.c#L135
- CSI data format description on the [README](https://github.com/seemoo-lab/nexmon_csi?tab=readme-ov-file#analyzing-the-csi)
- `int16` real and imaginary is only true for `bcm4339` and `bcm43455c0`, when read out they are int14, see conversion here: https://github.com/seemoo-lab/nexmon_csi/blob/fdb25ef0e4e1402e968bb644d4914ad1a3d0a84d/src/csi_extractor.c#L232
- `bcm4358` and `bcm4366c0` are in a floating format, see comments here: https://github.com/seemoo-lab/nexmon_csi/blob/fdb25ef0e4e1402e968bb644d4914ad1a3d0a84d/src/csi_extractor.c#L240

Urb1101, 2024-10-31:
thank you for nice information. but I only got rough info about format from README and already knew each CHIP has different processing by reading readcsi.m .
especially I want to know payload part format in detail for me to separate that each subcarrier, where can I get detail?
for example, I think that 'csi_buffer' in csireader.m before executing reshape(2,:) has this matrix↓
[subcarrier1 real, subcarrier2 real,... subcarrierN real, subcarrier1 imaginary, cubcarrier2 imaginary...subcarrierN imaginary] in each row, and ultimately that become 
[subcarrier1 real+imaginary, subcarrier2 real+imaginary,... subcarrierN real+imaginary] in each row,
and whether which is correct is What I want to know. 

I apologize for complex explaining.

jlinktu, 2024-10-31:
The code should be self-explaining..
`Hout` before `reshape` ([line 48](https://github.com/seemoo-lab/nexmon_csi/blob/fdb25ef0e4e1402e968bb644d4914ad1a3d0a84d/utils/matlab/csireader.m#L48)) is a one-dimensional vector holding interleaved real and imaginary parts for each subcarrier. After `reshape`, `Hout` is two-dimensional, where the first dimension holds real parts over all subcarriers and the sceond dimension holds respective imaginary parts. Hence, each row resembles one subcarrier. In [line 49](https://github.com/seemoo-lab/nexmon_csi/blob/fdb25ef0e4e1402e968bb644d4914ad1a3d0a84d/utils/matlab/csireader.m#L49), real and imaginary parts are combined to form a complex value for each ob the subcarriers.
In [line 50](https://github.com/seemoo-lab/nexmon_csi/blob/fdb25ef0e4e1402e968bb644d4914ad1a3d0a84d/utils/matlab/csireader.m#L50), the complex values of all subcarrier are assigned to one row of `csi_buff`.
Therefore, after executing the script, each row of `csi_buff` holds complex values of one extraction, where each columns represents one subcarrier.

Urb1101, 2024-10-31:
your explaining is almost same as mine and thats what I wanted to check. thank you for your corresponding!
im glad if you let me know where  "a one-dimensional vector holding interleaved real and imaginary parts for each subcarrier" generate.(thats ok if possible)

closed 2024-10-31 by Urb1101

jlinktu, 2024-10-31:
> im glad if you let me know where "a one-dimensional vector holding interleaved real and imaginary parts for each subcarrier" generate.(thats ok if possible)  

For `bcm4339` and `bcm43455c0` this is done in [line 39](https://github.com/seemoo-lab/nexmon_csi/blob/fdb25ef0e4e1402e968bb644d4914ad1a3d0a84d/utils/matlab/csireader.m#L39) by typecasting the payload into `int16` values. For `bcm4358` and `bcm4366c0` it is done by converting the payload from floating point format to integer values (line [41](https://github.com/seemoo-lab/nexmon_csi/blob/fdb25ef0e4e1402e968bb644d4914ad1a3d0a84d/utils/matlab/csireader.m#L41) and [43](https://github.com/seemoo-lab/nexmon_csi/blob/fdb25ef0e4e1402e968bb644d4914ad1a3d0a84d/utils/matlab/csireader.m#L43)). The conversion magic happens in [`unpack_float.c`](https://github.com/seemoo-lab/nexmon_csi/blob/master/utils/matlab/unpack_float.c).

If you wan't to understand this process in more detail please go through the respective code yourself, everything is opens-source and can be inspected by you.