seemoo-lab/nexmon_csi #369: CSI date loss

yjxb1, 2024-11-02:
Use a Raspberry Pi as the sender, ping data packets to the router at an interval of 0.001, and then use another Raspberry Pi to collect CSI data, but encountered problems during the collection. I capture CSI packets through tcpdump and store 1000 CSI packets as a pcap file. When I view the timestamp of the last CSI data in the first pcap file and the first time in the second pcap file When poking, it was found that the difference between the two was 0.7S, which means that 0.7S of CSI data was lost, as shown in the picture below. I'm confused, does anyone know why? Or how to solve this problem?
[image]
[image]

jlinktu, 2024-11-02:
Do I understand correctly that you are executing `tcpdump` to capture 1000 packets twice and are wondering why there is a 0.7 second long gap between the two invocations of `tcpdump`?
I would assume that this is the time it takes the first instance of `tcpdump` to finish its process after collecting 1000 packets and it takes the second instance of `tcpdump` to be in a state where it can start fetching packets off the interface.
That doesn't seem to be an issue with the Nexmon CSI extractor. For more information on how to use `tcpdump` I suggest to check out [its manpage](https://www.tcpdump.org/manpages/tcpdump.1.html).

yjxb1, 2024-11-03:
> Do I understand correctly that you are executing `tcpdump` to capture 1000 packets twice and are wondering why there is a 0.7 second long gap between the two invocations of `tcpdump`? I would assume [...]

Thanks for your answer. Is there any way to collect continuous data? I need to save 1000 data packets as a pcap file, and there is no 0.7S data gap between two data packets.

jlinktu, 2024-11-03:
Here are three suggestions that you can try:
1. Capture with `tcpdump` in a single session and split the capture file into junks of 1000 packets afterwards. E.g. using `editcap`: `editcap -c 1000 input.pcap output.pcap`. (see [`editcap` manual page](https://www.wireshark.org/docs/man-pages/editcap.html))
2. Pre-calculate the size of a `pcap` file of 1000 packets and use `tcpdump`'s `-C file_size` option to let `tcpdump` write to a new file every 1000 packets. (see [`tcpdump` manual page](https://www.tcpdump.org/manpages/tcpdump.1.html))
3. Write an own tool that listens on the interface and writes packets to files as required. You might wan't to have a look at [`libpcap`](https://www.tcpdump.org/manpages/pcap.3pcap.html).  

As this issue is not related to the extractor I am closing it as completed.

closed 2024-11-03 by jlinktu