seemoo-lab/nexmon_csi #371: About nexutil parameters

maple-42, 2024-11-18:
Hello.
Sorry if this is a rudimentary question or if it is mentioned somewhere else.

nexutil -Iwlan0 -s500 -b -l34 -vm+IBEQGIAgAAESIzRFWqu6q7qrsAAAAAAAAAAAAAAAAAAA==

What does the -s stand for in this code?
I had always thought that -s500 meant a sampling rate of 500 milliseconds (2Hz).
When I tried -s10 (100Hz), wireshark showed a sampling rate of 208.7Hz.

If my understanding is incorrect, I would appreciate it if you could let me know.

Thank you in advance.

jlinktu, 2024-11-18:
[`nexutil`](https://github.com/seemoo-lab/nexmon/tree/master/utilities/nexutil) provides a help message containing explanations of its parameters, you can get it by executing `nexutil --help`. From there you can see that `-s` takes as argument an IOCTL cmd id. This has absolutely nothing to do with a sampling rate.  

The IOCTL cmd of id 500 is handling the configuration of the CSI extractor. The corresponding handler in the firmware can be found [here](https://github.com/seemoo-lab/nexmon_csi/blob/fdb25ef0e4e1402e968bb644d4914ad1a3d0a84d/src/ioctl.c#L125).

closed 2024-11-18 by jlinktu

maple-42, 2024-11-18:
Sorry, I could not clear up the question by looking at it.

jlinktu, 2024-11-18:
It says:  
> How to control the extraction rate?
> As the CSI extractor works per received Wi-Fi frame the rate can be controlled by the transmitter.

maple-42, 2024-11-18:
Does that mean that a special command is needed or something, or is it a transmitter setting?

maple-42, 2024-11-18:
Two ASUS RT-AC86Us are used.

jlinktu, 2024-11-18:
The CSI extractor will extract CSI each time it overhears a Wi-Fi frame that matches its configured criteria. Thus, it has no control over the rate at all.

If the transmitter is part of a legitimate network, the rate of transmitted frames can have tons of conditions and isn't that simple to control. One can play with `iperf` and its payload sizes to affect the transmission rate, but you won't really have control over the timing of the Wi-Fi frames.
If instead injecting frames on the transmitter, you can have more control over the timing. If you are using an Asus RT-AC86U I suggest to use [`nexmon_tx_task`](https://github.com/seemoo-lab/nexmon_tx_task) to do this. There you have control over the time delay in between consecutive transmissions.