seemoo-lab/nexmon_csi #373: Estimating Maximum Signal Range for Nexmon CSI on Raspberry Pi

Asafe33, 2024-11-29:
Hello everyone, I’ve been researching Nexmon CSI and Wi-Fi for a while, and I’d like to thank the developers for this amazing tool. However, I have a question: what is the maximum distance the Raspberry Pi can receive signals from the router and collect data using Nexmon? Is there a formula to estimate this? I’ve read through some articles with many formulas, but I couldn’t find anything in the repository that directly answers this question. There are discussions about AoA and ToF, but these don’t seem to apply to the Raspberry Pi. Does anyone have an answer?

jlinktu, 2024-11-29:
The signal range is mainly influenced by a transmitter's transmit power. The receiving end, which a CSI extractor is, can only process a signal that reaches it. There are possibilities to amplify weak signals, e.g. by special antennas or front-ends, but if no energy arrives at the receiver there is nothing you can do.
TLDR: Transmitter influences the range, not the receiver.

Asafe33, 2024-11-29:
Let me see if I understand this correctly: if I can’t determine the maximum distance the Raspberry Pi can extract data, I can instead infer the maximum distance the home router can transmit and adjust that to my work, right? So, how can I estimate the maximum signal power of my home router in a line-of-sight scenario without obstacles? Is it given by the formula friss? And another question what is the sensitivity of the receiver (RPI) for signal reception?