seemoo-lab/nexmon_csi #374: CSI Extraction in Wi-Fi Client Mode for bcm43455c0

munir01, 2024-12-12:
Hello everyone,

Thank you for the amazing work on nexmon_csi—it has been incredibly valuable for the community!

I have a quick question: Is it possible to extract CSI in Wi-Fi client mode on the bcm43455c0 chipset, rather than using monitor mode? If so, are there any specific modifications required?

Looking forward to any insights. Thanks!

Best,
Munir

zeroby0, 2024-12-13:
The radio is 'deafed' when transfering CSI to the kernel to prevent it being overwritten while transferring. So some packets in transmission will never be seen by the radio. Also, the way the CSI data is transferred is by overwriting the metadata in the packet. So you can't have the pi connected while collecting CSI.

Maybe you can have a dongle, not uninstall wpa_supplicant, and get it working. It would be, in my opinion, lots of effort and less stable.

munir01, 2024-12-13:
Thank you for your response—it was very helpful!

Do you think switching between monitor mode and Wi-Fi client mode for occasional packet transmission would cause instability or other issues? Additionally, how much delay is typically expected when switching between these modes?

Thanks again!