seemoo-lab/nexmon_csi #378: CSI Collected Data Differences Across Builds 5.10.92 and 5.10.103

ML34, 2025-03-28:
Hi everyone,

I've been testing CSI data collection using Nexmon on a Raspberry Pi 4B and noticed significant differences in CSI packet timing and amplitude stability between two setups, even though all hardware and collection parameters were kept the same.

Setup:
Device: Raspberry Pi 4B , Wi-Fi Chipset: BCM43455

CSI Tool:

SD#1: Pre-compiled Nexmon binaries (https://github.com/nexmonster/nexmon_csi_bin) with Kernel 5.10.92

SD#2: Nexmon CSI built from source (https://github.com/seemoo-lab/nexmon_csi) with Kernel 5.10.103

Same AP and MAC address used

Same CSI collection parameters and pcap capture method (tcpdump)

Same number of packets collected (~5000)

Observation:
CSI amplitude plots from SD#2 are more stable with fewer fluctuations and spikes.
No obvious packet loss in either case — the number of packets is identical.

Question:
Could differences in the build process, Nexmon version, or kernel minor versions affect the sampling rate or driver behavior?

I'm happy to share plots or packet timing data if helpful. Thanks for your work on this project — it's been a great tool for our research!

jlinktu, 2025-03-28:
CSI is extracted whenever a WLAN frame is processed that matches the configured filter criteria. Once the CSI is extracted it is wrapped in a UDP packet, which is then forwarded to the host device. There is no sampling rate as there is no sampling.

The **magnitude** of CSI can be affected by various factors, such as wireless channel's effects, hardware components, precoding, transmit diversity schemes... If you don't control the transmitter you can't rule out that the effects you are observing are related to the transmitting device.

We do not provide any precompiled / prepatched firmware and also advice against using those. You should always generate your own patched firmware.

ML34, 2025-03-28:
Thank you for the clarification — that helps a lot in understanding the CSI extraction process and the factors involved.

In our case, we kept the setup, environment, and all parameters exactly the same across both scenarios: same Raspberry Pi 4B, same AP, same bandwidth/channel, and same traffic pattern (constant ICMP PINGs at 100 Hz).

Despite this, we noticed that the CSI amplitude data shows noticeably different stability between the two setups. One setup consistently yields smoother amplitude trends across subcarriers, with fewer spikes or fluctuations, while the other shows more variance and occasional anomalies.

We're aware that channel effects, hardware variability, and transmitter behavior can influence CSI magnitudes. However, since all external conditions and parameters are identical, we're exploring whether subtle differences in firmware builds or driver behavior might explain the observed mismatch.

We appreciate the recommendation to avoid precompiled firmware — we'll continue using self-built patched firmware from the source for better consistency.

Thanks again for your support and for maintaining such a powerful tool for the research community!

closed 2025-04-09 by jlinktu