seemoo-lab/nexmon_csi #388: Support for Raspberry Pi 5

pmssantos, 2025-08-06:
Hi all, 

Is it possible to update Nexmon CSI to be supported by a Raspberry Pi 5-compatible kernel?
I see the Nexmon repo has been updated to support kernel 6.6 (https://github.com/seemoo-lab/nexmon/issues/599), but not nexmon_csi.
Is this something I can do on my end?

Thanks

jlinktu, 2025-08-06:
See #368 
Additionally you might want to have a look on how to properly load a modified firmware on recent Raspberry Pi OS version (see https://github.com/seemoo-lab/nexmon/blob/92f7feac42df460690a94a9bf861642304e2929a/patches/bcm43455c0/7_45_234_4ca95bb_CY/nexmon/Makefile#L190)
Combining those hints you can definitely do it on your end

closed 2025-08-06 by jlinktu

pmssantos, 2025-08-07:
Ok, I looked around in documentation and Issues, but remain puzzled.
Setup: RPI v5, bcm43455c0, kernel 6.6.51+rpt-rpi-v8, using **patches/bcm43455c0/7_45_234_4ca95bb_CY**; end goal is to extract CSI

- I read [#368](https://github.com/seemoo-lab/nexmon_csi/issues/368), but as far as I can see, it is of no immediate use to me; I am not even past a successful make of nexmon_csi (meanwhile, Nexmon compiles and installs the firmware fine, as long as you skip **apt-get upgrade**, as that will upgrade your kernel from 6.6.y). While performing point 10 in the [Nexmon_csi instructions](https://github.com/seemoo-lab/nexmon_csi?tab=readme-ov-file#bcm43455c0), which essentially is
`./nexmon/patches/bcm43455c0/7_45_234_4ca95bb_CY/nexmon_csi# make` 
compilation errors occur:

```
src/csi_extractor.c:331:51: error: invalid type argument of unary '*' (have 'int')
   GenericPatch1(initvals_rxhdr_len0, (RXE_RXHDR_LEN * 2));
                                                     ^
/home/p/nexmon/patches/include/patcher.h:66:51: note: in definition of macro 'GenericPatch1'
     unsigned char gp1_ ## name = (unsigned char) (val);
```
[Link to csi_extractor.c](https://github.com/seemoo-lab/nexmon_csi/blob/fdb25ef0e4e1402e968bb644d4914ad1a3d0a84d/src/csi_extractor.c#L331)
Ok, I look into it (already reported in [#316 ](https://github.com/seemoo-lab/nexmon_csi/issues/316)). **GenericPatchX** are some macros defined in **nexmon/patches/include/patcher.h** that are not receving RXE_RXHDR_LEN as input. Sure enough, RXE_RXHDR_LEN is not defined in **definitions.mk** (recall, I am using 7_45_234_4ca95bb_CY; they are defined in e.g. [7_45_206](https://github.com/seemoo-lab/nexmon/blob/92f7feac42df460690a94a9bf861642304e2929a/firmwares/bcm43455c0/7_45_206/definitions.mk#L5)), so I add it and problem gets solved (quick and dirty, I know; but I just wanted to see some progress). Same thing for VERSION_PTR (next error, similar issue). But now I am stuck at 
`make: *** No rule to make target 'obj/ucode_compressed.o', needed by 'gen/nexmon2.pre'.  Stop.`
Needless to say, **make install-firmware** also fails with
`make: *** No rule to make target 'brcmfmac43455-sdio.bin', needed by 'install-firmware'.  Stop.`

- As for [https://github.com/seemoo-lab/nexmon/blob/92f7feac42df460690a94a9bf861642304e2929a/patches/bcm43455c0/7_45_234_4ca95bb_CY/nexmon/Makefile#L190]( https://github.com/seemoo-lab/nexmon/blob/92f7feac42df460690a94a9bf861642304e2929a/patches/bcm43455c0/7_45_234_4ca95bb_CY/nexmon/Makefile#L190): I get it (or I think I do), **make install-firmware** will apply the patched firmware. But inspecting the makefile for **nexmon_csi** (and not the nexmon Makefile), I see a dependency on [kernel versions](https://github.com/seemoo-lab/nexmon_csi/blob/fdb25ef0e4e1402e968bb644d4914ad1a3d0a84d/Makefile#L99):

```
# only make for bcm43455c0  
ifneq ($(findstring bcm43455c0,$(FWUCODE)), )  
brcmfmac.ko: check-nexmon-setup-env
ifeq ($(shell uname -m),$(filter $(shell uname -m), armv6l armv7l))
ifeq ($(findstring 4.19,$(shell uname -r)),4.19)
        @printf "\033[0;31m  BUILDING DRIVER for kernel 4.19\033[0m brcmfmac_4.19.y-nexmon/brcmfmac.ko (details: log/driver.log)\n" $@
        $(Q)make -C /lib/modules/$(shell uname -r)/build M=$$PWD/brcmfmac_4.19.y-nexmon -j2 >log/driver.log
else ifeq ($(findstring 5.4,$(shell uname -r)),5.4)
        @printf "\033[0;31m  BUILDING DRIVER for kernel 5.4\033[0m brcmfmac_5.4.y-nexmon/brcmfmac.ko (details: log/driver.log)\n" $@
        $(Q)make -C /lib/modules/$(shell uname -r)/build M=$$PWD/brcmfmac_5.4.y-nexmon -j2 >log/driver.log
else ifeq ($(findstring 5.10,$(shell uname -r)),5.10)
        @printf "\033[0;31m  BUILDING DRIVER for kernel 5.10\033[0m brcmfmac_5.10.y-nexmon/brcmfmac.ko (details: log/driver.log)\n" $@
        $(Q)make --trace -C /lib/modules/$(shell uname -r)/build M=$$PWD/brcmfmac_5.10.y-nexmon -j1 # >log/driver.log
else
        $(warning Warning: Kernel version not supported)
endif
```

I really don't mind continuing to look around, but I'd like to understand if this is the right direction or not, or if there's something obvious that I am missing (e.g., is the nexmon_csi firmware really necessary, or is nexmon sufficient for CSI extraction?).
I'll continue to press on, but any help is welcome.
Thanks